acshk/acsccid #2: Segmentation fault during initialize

LoicGomez, 2015-12-08:
Hi,

I've been trying to initialize an ACS Cryptomate64 token using this driver on Ubuntu wily 64 bits (precompiled libacsccid1_1.1.1-1~ubuntu15.10.1_amd64.deb and compiled from the sources) but it fails with SIGSEGV.

I've attached the debug log.

Could you help me ?
Thanks

[cryptomate64-debug.txt](https://github.com/acshk/acsccid/files/55188/cryptomate64-debug.txt)

godfreychung, 2015-12-08:
Hi LoicGomez

According to the log, it seems that OpenSC does not support ACOS5 card.

Did OpenSC fail with SIGSEGV?

LoicGomez, 2015-12-10:
Hi,

That is strange, it should load the module from your packages, right ?

This is what I get in syslog when inserting the card, pkcs11-tool is segfaulting :
> kernel: [ 7366.751210] usb 1-2: new full-speed USB device number 11 using xhci_hcd
> kernel: [ 7366.882418] usb 1-2: New USB device found, idVendor=072f, idProduct=90db
> kernel: [ 7366.882422] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> kernel: [ 7366.882424] usb 1-2: Product: CryptoMate64
> kernel: [ 7366.882427] usb 1-2: Manufacturer: ACS
> mtp-probe: checking bus 1, device 11: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2"
> mtp-probe: bus: 1, device: 11 was not an MTP device

And when trying to initialize the token :
> kernel: [ 7381.615775] xhci_hcd 0000:00:14.0: WARN Event TRB for slot 10 ep 4 with no TDs queued?
> (repeats 12 times)
> kernel: [ 7381.710117] pkcs11-tool[8546]: segfault at 10 ip 00007f32fecb8af2 sp 00007fff28241f60 error 4 in opensc-pkcs11.so[7f32fecaf000+2c000]

godfreychung, 2015-12-10:
Hi LoicGomez

When you plug CryptoMate64 to your PC, pcscd will recognize the device and load acsccid. You can run pcscd in debug mode to log the messages.

Please kill any running pcscd process before running the following command.
sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd | tee log.txt

LoicGomez, 2015-12-11:
Thanks, here is the log with insertion + pkcs11-tool init attempt
[cryptomate64-pcscd.txt](https://github.com/acshk/acsccid/files/59171/cryptomate64-pcscd.txt)

.

godfreychung, 2015-12-11:
According to the log, the driver was working properly.

pkcs11-tool sent 00 A4 00 00 02 3F 00 to CryptoMate64 but 69 86 was returned.
I think CryptoMate64 is not initialized properly so that the file does not exist.

LoicGomez, 2015-12-11:
Yeah but the command I launch was in fact the initialize as any other command is replying "token not initialized"
$ pkcs11-tool --init-token --so-pin 0000 --label testlabel

godfreychung, 2015-12-12:
MF (File ID: 3F 00) was not found. Therefore, the device did not personalize with PKCS #15.

LoicGomez, 2015-12-14:
I'm sorry I don't understand what you mean, could you please elaborate ? I'm not sending files or using files in the command line, is it maybe internal ?

Thank you very much

godfreychung, 2015-12-14:
I mean Master file (MF) on the smart card. It should follow PKCS #15 format.

LoicGomez, 2015-12-14:
I see, but I don't really understand, this is a driver specifically for these [Cryptomate64](http://www.acs.com.hk/en/products/18/cryptomate64-cryptographic-usb-tokens/) cards right ? So it should have a master file following the PKCS#15 format.

ACS support tells me the issue lies in the fact OpenSC does not support ACOS5 cards and Cryptomate64 is an ACOS5. But then I don't understand what tool I am supposed to use to initialize and effectively use this smartcard+reader.

I hope I'm making sense here ^^

godfreychung, 2015-12-14:
CryptoMate64 is a smart card reader with ACOS5 card. The contents of the card is empty and you should personalize it to use for any purpose. For OpenSC, it works with the card with PKCS #15 format.

Yes, OpenSC does not support ACOS5 card yet and it cannot personalize the card to PKCS #15 format.

I think you should take a look on ACS CryptoMate64 Client Kit.

LoicGomez, 2015-12-14:
I don't understand, isn't this driver supposed to replace ACS Cryptomate64 client kit for Linux ? We are using Linux only, we can't use ACS CryptoMate64 Client Kit.

Is there another option for Linux ? ACS is advertising Linux support and I don't see any right now since there is a driver (acsccid) but no compatible tool... Could you point to Linux compatible tools to manage these smartcards with your project ?

godfreychung, 2015-12-15:
No, you can use the tool from the client kit.
For the platform support, please use our latest acsccid driver.

LoicGomez, 2015-12-16:
Ok, will do, thank you for your help.

closed 2015-12-16 by LoicGomez