acshk/acsccid #3: Since I disabled the pn533, the LED and Buzzer on the ACR122U doesn't work.

comseong, 2016-03-23:
I am using acr122u-a9 at Raspberry-pi 2 with Raspbian. 
When the ACR122U is plugged in, kernel(>3.5) will automatically load the pn533 driver. With the pn533 driver, pcscd will report "Can't claim interface" error. When I disabled pn533 and nfc driver in kernel with blacklist.conf, the error will be disappeared.
But since I disabled the pn533, the LED and Buzzer on the ACR122U doesn't work. 
How can I solve this problem?
Thanks.

godfreychung, 2016-03-23:
I don't think it is related to pn533 kernel driver.
Did you try to send the escape command to control the LED and the buzzer?

comseong, 2016-03-29:
The default value of both of LED and buzzer should be ON without any control command. When I plug in the ACR122U to other OS like Windows, the LED and buzzer works well without any configuration. When I enabled the pn533 driver, both of LED and buzzer will work again.

godfreychung, 2016-03-30:
Can you provide ACR122U firmware version?

comseong, 2016-04-01:
ACR122U215

godfreychung, 2016-04-01:
It seems that the driver was not loaded.
Did you run pcscd with acsccd driver?
Was the reader able to detect the card from pcsc_scan?

comseong, 2016-04-04:
I installed this acsccid by source code compile from this gitbub repository. The moment when I execute the pcsc_scan command, the Buzzer and LED turns ON. Do I always have to use the command to turn ON the Buzzeer and LED? But when I stop the pcsc_scan demon, the Buzzer and LED will not work again. Is it normal?

godfreychung, 2016-04-05:
Yes, it is normal. pcscd is controlled by systemd in Raspbian Jessie. systemd will start pcscd if a PC/SC application access the reader. Otherwise, it will stop pcscd.

Therefore, I think you will run you own PC/SC application to access the card.

comseong, 2016-04-05:
I used wrong expression. The pcsc_scan is not a demon but application. When I stop the pcsc_scan application forcibly by CTRL+C, the Buzzer and LED will not work again. You mean that turning ON the Buzzer and LED is responsible for application? But when I load the PN533 kernel driver on the Linux and I plugged in the reader to Windows OS, the Buzzer and LED will work automatically without executing any application.

godfreychung, 2016-04-05:
As I mentioned before, pcscd is controlled by systemd in Raspbian Jessie. When pcsc_scan accesses a reader, it will trigger systemd to start pcscd and then pcscd loads acsccid driver for ACR122U. However, systemd will stop pcscd if there is no application accessing the reader. Therefore, the buzzer and LED is not working because acsccid driver is unloaded.

For Linux kernel driver pn533 or Windows driver, the driver is running until the reader is unplugged.

comseong, 2016-04-05:
I understood that the pcscd should be loaded statically by application. But is there any method for the pcscd to be loaded dynamically when I plug the reader in Raspbian like a plug-n-play? Except for me, the other people who want to do a Raspberry-Pi project with this reader are also confusing about this.

godfreychung, 2016-04-05:
I think you can switch systemd to sysvinit in Raspbian Jessie.

comseong, 2016-04-06:
Thank you for your help!

comseong, 2016-04-06:
Unfortunately other problem has happened. Since I have started the pcscd manually by "service pcscd start" command, the Buzzer and LED work well when I tagged a card. But when I use the nfc-tool command like nfc-list or lsnfc, the Buzzer and LED doesn't work again even though the pcscd is running on the memory. Can't I use both the pcscd and libnfc at the same time?

godfreychung, 2016-04-07:
I think libnfc didn't restore the default card polling behavior after using the reader.

godfreychung, 2016-04-14:
This issue was solved by switching systemd to sysvinit but the libnfc should restore the default card polling behavior after using the reader.

closed 2016-04-14 by godfreychung