acshk/acsccid #4: Extra slot on ACR1281 1S

nmikhailov, 2016-04-23:
My ACR1281 1S Dual Reader vid: 072f pid: 2224 reports invalid number of slots:
```
`% pcsc_scan
PC/SC device scanner
V 1.4.26 (c) 2001-2011, Ludovic Rousseau <ludovic.rousseau@free.fr>
Compiled with PC/SC lite version: 1.8.16
Using reader plug'n play mechanism
Scanning present readers...
0: ACS ACR1281 1S Dual Reader 00 00
1: ACS ACR1281 1S Dual Reader 00 01
2: ACS ACR1281 1S Dual Reader 00 02

Sat Apr 23 22:33:45 2016
Reader 0: ACS ACR1281 1S Dual Reader 00 00
  Card state: Card removed,
Reader 1: ACS ACR1281 1S Dual Reader 00 01
  Card state: Card removed,
Reader 2: ACS ACR1281 1S Dual Reader 00 02
  Card state: Card inserted, Unresponsive card,
```

Note reader 2 which is always in state `Card inserted, Unresponsive card` which is annoying

godfreychung, 2016-04-24:
The result is normal. The 3rd slot is a SAM slot in the reader.

nmikhailov, 2016-04-24:
ok, thanks for the info

closed 2016-04-24 by nmikhailov