acshk/acsccid #6: Question regarding SAM reader

FacundoMainere, 2017-07-07:
Hello there, we have one of your readers(ACR1252U) and the example codes from the SDK. We inserted the SAM and tried the ACOS6 java example. When we press the mutual authenthication button we receive the error code "6B 80"  when the code is trying to retrieve the Serial number of the card, but we couldnt identify what that error means(["Reference incorrect (procedure byte)" according to this](https://eftlab.co.uk/index.php/site-map/knowledge-base/118-apdu-response-list)).
could you clarify whats wrong please? is it an issue with this driver?

godfreychung, 2017-07-07:
You may need to run pcscd in foreground debug mode to see if there is CCID error from the reader.

FacundoMainere, 2017-07-07:
This is the log when it tries to do the get serial operation

> 14556563 winscard_svc.c:359:ContextThread() Received command: TRANSMIT from client 15
> 00000018 readerfactory.c:852:RFReaderInfoById() RefReader() count was: 1
> 00000003 winscard.c:1605:SCardTransmit() Send Protocol: T=1
> 00000003 APDU: 80 14 00 00 08 
> 00000003 ifdhandler.c:1551:IFDHTransmitToICC() usb:072f/223b:libudev:0:/dev/bus/usb/003/006 (lun: 0)
> 00000418 commands.c:1589:CCID_Receive Activity aborted by Host
> 00000012 SW: 6B 80 
> 00000004 winscard.c:1650:SCardTransmit() UnrefReader() count was: 2
> 00000004 winscard_svc.c:672:ContextThread() TRANSMIT rv=0x0 for client 15

estefaniamiguel, 2017-07-10:
I have the same error

josx, 2017-07-10:
Same error here, 
I cant find a working code.

leo-orellana, 2017-07-10:
I have the same problem.
This model does not appear in the README in the list of supported readers

godfreychung, 2017-07-10:
@FacundoMainere According to log messages, it seems to be firmware issue. Do you know firmware version of ACR1252U-M1?

FacundoMainere, 2017-07-10:
@godfreychung seems to be ACR1252U_V104.6

godfreychung, 2017-07-12:
@FacundoMainere Your reader is running the latest firmware. According to log messages, you connected the card using T=1 but ACOS6 is using T=0 only. Therefore, pcsc-lite will reject the connection. I tried it using pcsc-lite 1.8.22 and found that pcsc-lite will reject the connection if I select T=1 for connection.

Can you generate the log using the following command?
sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd | tee log.txt

FacundoMainere, 2017-07-12:
Hey @godfreychung , thanks for the feedback. By changing the T we could establish the connection and get the serial number in a python script. The java example in the SDK doesn't work with the reader out of the box so it would be good to correct it with this too.

closed 2017-07-12 by FacundoMainere