acshk/acsccid #9: ACR39U-NF support (USB-C reader)

pimlottc-gov, 2017-10-16:
I'm having problems using the new compact USB-C reader on my new MacBook Pro 2017.  Testing with Centrify Smart Card Assistant shows the following PKI errors:

```
** Data signing failed: CSSM_DecryptData failed: CSSMERR_CSP_INTERNAL_ERROR 
** Signature verification failed: CSSM_EncryptData failed: CSSMERR_CSP_INPUT_LENGTH_ERROR 
Public key encryption succeeded 
** Private key decryption failed: CSSM_DecryptData failed: CSSMERR_CSP_INTERNAL_ERROR 
** Private key encryption failed: CSSM_DecryptData failed: CSSMERR_CSP_INTERNAL_ERROR 
** Public key decryption failed: Unknown PKCS#1 padding type 0x98 
```

In contrast, the same card works fine using the older USB-A reader (ACR38U-N1) and a USB-A to C adapter:

```
         Signature verification succeeded
         Public key encryption succeeded
         Private key decryption succeeded
         Decrypted data matched original
         Private key encryption succeeded
         Public key decryption succeeded
         Decrypted data matched original
```

godfreychung, 2017-10-17:
I would like to know the ATR of your card.
Can you run pcsctest on your Mac?

pimlottc-gov, 2017-10-17:
I can run a pcsctest for you when I get back to my desk.  I do have the results of running the `parse` command from the ccid project:

```
 idVendor: 0x072F
  iManufacturer: ACS
 idProduct: 0xB100
  iProduct: ACR39U ICC Reader
 bcdDevice: 2.16 (firmware release?)
 bLength: 9
 bDescriptorType: 4
 bInterfaceNumber: 0
 bAlternateSetting: 0
 bNumEndpoints: 3
  bulk-IN, bulk-OUT and Interrupt-IN
 bInterfaceClass: 0x0B [Chip Card Interface Device Class (CCID)]
 bInterfaceSubClass: 0
 bInterfaceProtocol: 0
  bulk transfer, optional interrupt-IN (CCID)
 Can't get iInterface string
 CCID Class Descriptor
  bLength: 0x36
  bDescriptorType: 0x21
  bcdCCID: 1.10
  bMaxSlotIndex: 0x00
  bVoltageSupport: 0x07
   5.0V
   3.0V
   1.8V
  dwProtocols: 0x0000 0x0003
   T=0
   T=1
  dwDefaultClock: 4.800 MHz
  dwMaximumClock: 16.000 MHz
  bNumClockSupported: 0 (will use whatever is returned)
   IFD does not support GET CLOCK FREQUENCIES request: No such file or directory
  dwDataRate: 12903 bps
  dwMaxDataRate: 826000 bps
  bNumDataRatesSupported: 0 (will use whatever is returned)
   IFD does not support GET_DATA_RATES request: No such file or directory
  dwMaxIFSD: 247
  dwSynchProtocols: 0x00000000
  dwMechanical: 0x00000000
   No special characteristics
  dwFeatures: 0x000100B2
   ....02 Automatic parameter configuration based on ATR data
   ....10 Automatic ICC clock frequency change according to parameters
   ....20 Automatic baud rate change according to frequency and Fi, Di params
   ....80 Automatic PPS made by the CCID
   01.... TPDU level exchange
  dwMaxCCIDMessageLength: 271 bytes
  bClassGetResponse: 0xFF
   echoes the APDU class
  bClassEnvelope: 0xFF
   echoes the APDU class
  wLcdLayout: 0x0000
  bPINSupport: 0x00
  bMaxCCIDBusySlots: 1
```

pimlottc-gov, 2017-10-18:
Here's the `pcsctest` output:

```
$ pcsctest 

MUSCLE PC/SC Lite Test Program

Testing SCardEstablishContext    : Command successful.
Testing SCardGetStatusChange 
Please insert a working reader   : Command successful.
Testing SCardListReaders         : Command successful.
Reader 01: ACS ACR39U ICC Reader
Enter the reader number          : 01
Waiting for card insertion         
                                 : Command successful.
Testing SCardConnect             : Command successful.
Testing SCardStatus              : Command successful.
Current Reader Name              : ACS ACR39U ICC Reader
Current Reader State             : 0x54
Current Reader Protocol          : 0x0
Current Reader ATR Size          : 20 (0x14)
Current Reader ATR Value         : 3B 7F 96 00 00 80 31 80 65 B0 84 23 27 E5 12 0F FE 82 90 00 
Testing SCardDisconnect          : Command successful.
Testing SCardReleaseContext      : Command successful.
Testing SCardEstablishContext    : Command successful.
Testing SCardGetStatusChange 
Please insert a working reader   : Command successful.
Testing SCardListReaders         : Command successful.
Reader 01: ACS ACR39U ICC Reader
Enter the reader number          : 01
Waiting for card insertion         
                                 : Command successful.
Testing SCardConnect             : Command successful.
Testing SCardStatus              : Command successful.
Current Reader Name              : ACS ACR39U ICC Reader
Current Reader State             : 0x54
Current Reader Protocol          : 0x0
Current Reader ATR Size          : 20 (0x14)
Current Reader ATR Value         : 3B 7F 96 00 00 80 31 80 65 B0 84 23 27 E5 12 0F FE 82 90 00 
Testing SCardDisconnect          : Command successful.
Testing SCardReleaseContext      : Command successful.

PC/SC Test Completed Successfully !
```

godfreychung, 2017-10-18:
According to the ATR of your card, it is a T=0 card. During APDU transmission, the driver converts APDU to TPDU and submits it to the reader. I think that the firmware may have timing issue in processing the response from the card.

Did you have any problem in selecting file, reading records or other operations?

I assumed that you have installed acsccid driver. To check your current installation, you can go to /usr/local/libexec/SmartCardServices/drivers and see if ifd-acsccid.bundle exists or not.

If you have ACR39U or ACR39U-N1 on hand, you can test it with your card and compare the result.

pimlottc-gov, 2017-10-18:
Yes, I'm having problems.  We are using Centrify for PIV based login bound to AD.  Log in with these readers does not work; it detects the card and changes the login box to input PIN but the PIN is not accepted.

We do have a bunch of USB-A card readers (ACR38U-N1) that work fine, both for actual user login and the Centrify diagnostic test from my initial issue report (which is just running ```sctool -D```).  We also have been using Identiv SCR3500 readers with no problems.  Now we just need a USB-C solution for new MacBooks that don't have USB-A ports.

pimlottc-gov, 2017-10-18:
I am using acsccid 1.1.4

godfreychung, 2017-10-19:
For your case, it may be firmware or hardware issue. Please kindly contact our USA office (https://www.acs.com.hk/en/acs-worldwide/) to troubleshoot your problem.

godfreychung, 2017-12-11:
Did `acsccid_installer-1.1.5.1.dmg` solve your problem?

pimlottc-gov, 2017-12-15:
It does seem to resolve the issue so far @godfreychung.  The sctool test passes and I am able to login.

godfreychung, 2018-10-30:
Fixed in 8011c9682d36a69883cfd057b9ac1c16bee5c7ae.

closed 2018-10-30 by godfreychung