acshk/acsccid #10: Differentiate between SAM and PICC slots for ACR1252

ctadlock, 2017-11-07:
When using the ACR1252U, and testing on a Mac with the latest ACS PC/CS drivers installed, each NFC reader returns 2 reader names from `SCardListReaders` for each physical NFC reader. One of for the PICC slot, and one is for the SAM slot. However, it is not possible to determine which is PICC and which is SAM from just the PC/SC reader names. Is it always that slot 1 is PICC and slot 2 is SAM? We only want to work with the PICC.
 
// device 0
`ACS ACR1252 1S CL Reader(1)` // slot 1
`ACS ACR1252 1S CL Reader(2)` // slot 2
 
// device 1
`ACS ACR1252 1S CL Reader(1) 01` // slot 1
`ACS ACR1252 1S CL Reader(2) 01` // slot 2
 
Note that if the ACS PC/SC drivers are NOT installed, then it IS possible to determine which is the PICC and which is the SAM based on the reader name.
 
`ACS ACR1252 1S CL Reader PICC 0`
 
How can this be resolved? Seems like the best solution would be for the drivers to be updated to include PICC and SAM in the PC/SC reader name. If its possible for the default Apple drivers to do this, the the ACS drivers should also be able to distinguish between the two.

Note that that querying for a tag on the reader as a way to determine if the reader is SAM or PICC is not a workable solution. We need to be able to determine this WITHOUT having a tag on the reader.

godfreychung, 2017-11-08:
In the mean time, you can assume that slot 1 is PICC. Please update this issue if you can't use PICC in slot 1.

It is not possible to customize the reader name for each CCID interface. The reader name is specified in Info.plist for each VID/PID only.

ctadlock, 2017-11-11:
Slot 1 seems to be the PICC. Is this also true for the ACR1222? It seems so.

[image]

godfreychung, 2017-11-12:
Yes, the first slot is PICC and other slots are SAM.

ctadlock, 2017-11-12:
After much testing it looks like this isnt always true; at least on the ACR1252U. Here is a repro of where slot 2 is the PICC. So now Im back to my original question.. How can the PICC slot be determined without having a tag on the reader?

[image]

ctadlock, 2017-11-12:
If the driver is removed, only one reader is shown and its the PICC. So what is this driver doing that the native driver isn't?

[image]

ctadlock, 2017-11-12:
Is there an `SCARD_ATTR_?` that can be retrieved via `SCardGetAttrib` that would differentiate between the PICC and SAM?

godfreychung, 2017-11-13:
The native driver does not support ACR1252U as USB composite device. Therefore, only first CCID interface is supported.

Can you install the latest acsccid driver (i.e. v1.1.5) for testing?

ctadlock, 2017-12-01:
Follow up.. I havent had an issue where the PICC slot was #2 again. Seems like it was just that one time.

closed 2020-03-26 by godfreychung

ArtArt0, 2021-08-01:
I can produce that the 1st device is the SAM and the 2nd is the PICC! - fast reconnect and U will see!

So how can I detect the PICC correctly?