acshk/acsccid #14: Protocol T=1 not supported on ACR38U-CCID

Jookia, 2018-12-20:
I have an ACR38U-CCID (072f:90cc), running git master asccid, pcsc 1.8.20, with the ACOSJ card (ATR 3B69000241434F534A76313031).

The documentation says the card should support T=1, but on Linux and Windows using OpenSC it always fails. pcscd gives the error 'Protocol T=1 requested but unsupported by the card'. I spent a bit bypassing ATR checks in case the ATR was lying, but this too fails just with different errors.

godfreychung, 2018-12-24:
You need to customize the ATR in order to use T=1 protocol. For more information, please refer to ACOSJ reference manual.

warface75, 2019-06-25:
do you have the reference manual? or do you know where to find it? i've been looking for a while but i can't find it
thanks in advance

Jookia, 2019-06-25:
I have a copy of the reference manual. Apparently you have to request it by email from ACS, but I asked the vendor I bought my card from (Smartcard Focus) and they sent me a copy. The name is 'Reference Guide-ACOSJ-Contact-1.04.pdf'

warface75, 2019-06-26:
hi, can you send me a copy? because i bought my cards in another site but they doesn't sent me the manual
how can i get it? can you help me?

warface75, 2019-06-26:
my email es warface75 in gmail so i hope that you can help me please
thanks a lot

closed 2020-03-26 by godfreychung