acshk/acsccid #15: CCID unknown when updating file in smartcard (extended APDU)

amportugal, 2019-01-03:
When performing an update on a file with a 32Kb APDU (extended APDU), the following error is shown on pcscd logs:

    ...
    pcscd[14964]: 00000000 commands.c:1523:CCID_Receive Unknown CCID error: 0xC3
    pcscd[14964]: 00000029 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
    pcscd[14964]: 00000006 winscard.c:1626:SCardTransmit() Card not transacted: 0x80100016
    ...

And the card capabilities are as follows:

      Tag: 7, len: 3 (card capabilities)
      Selection methods: 84
        - DF selection by full DF name
        - Short EF identifier supported
      Data coding byte: 01
        - Behaviour of write functions: one-time write
        - Value 'FF' for the first byte of BER-TLV tag fields: invalid
        - Data unit in quartets: 2
      Command chaining, length fields and logical channels: E0
        - Command chaining
        - Extended Lc and Le fields
        - RFU (should not happen)
        - Logical channel number assignment: No logical channel
        - Maximum number of logical channels: 1

Which means the card supports extended APDUs. I also have the NFC reader documentation and it supports extended Lc and Le fields.

Requested information:

    ACS CCID Driver version: 1.1.6-1
    PCSClite version: 1.8.24-1

Output of the command `/usr/sbin/pcscd --version`:

    pcsc-lite version 1.8.24.
    Copyright (C) 1999-2002 by David Corcoran <corcoran@musclecard.com>.
    Copyright (C) 2001-2015 by Ludovic Rousseau <ludovic.rousseau@free.fr>.
    Copyright (C) 2003-2004 by Damien Sauveron <sauveron@labri.fr>.
    Report bugs to <pcsclite-muscle@lists.alioth.debian.org>.
    Enabled features: Linux x86_64-pc-linux-gnu libsystemd serial usb libudev usbdropdir=/usr/lib/pcsc/drivers ipcdir=/var/run/pcscd configdir=/etc/reader.conf.d

OS:

     Linux 4.15.0-20-generic #21-Ubuntu SMP Tue Apr 24 06:16:15 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

Standard compatible PC with following hardware:

    H/W path       Device      Class          Description
    =====================================================
                               system         Computer
    /0                         bus            Motherboard
    /0/0                       memory         7854MiB System memory
    /0/1                       processor      Intel(R) Core(TM) i5-3230M CPU @ 2.60GHz
    /0/100                     bridge         3rd Gen Core processor DRAM Controller
    /0/100/2                   display        3rd Gen Core processor Graphics Controller
    /0/100/14                  bus            7 Series/C210 Series Chipset Family USB xHCI Host Controller
    /0/100/16                  communication  7 Series/C216 Chipset Family MEI Controller #1
    /0/100/16.3                communication  7 Series/C210 Series Chipset Family KT Controller
    /0/100/19      enp0s25     network        82579LM Gigabit Network Connection
    /0/100/1a                  bus            7 Series/C216 Chipset Family USB Enhanced Host Controller #2
    /0/100/1b                  multimedia     7 Series/C216 Chipset Family High Definition Audio Controller
    /0/100/1c                  bridge         7 Series/C216 Chipset Family PCI Express Root Port 1
    /0/100/1c/0                generic        MMC/SD Host Controller
    /0/100/1c.1                bridge         7 Series/C210 Series Chipset Family PCI Express Root Port 2
    /0/100/1c.2                bridge         7 Series/C210 Series Chipset Family PCI Express Root Port 3
    /0/100/1c.2/0  wlp4s0      network        Centrino Wireless-N 2230
    /0/100/1c.5                bridge         7 Series/C210 Series Chipset Family PCI Express Root Port 6
    /0/100/1c.5/0              bus            uPD720200 USB 3.0 Host Controller
    /0/100/1d                  bus            7 Series/C216 Chipset Family USB Enhanced Host Controller #1
    /0/100/1f                  bridge         QM77 Express Chipset LPC Controller
    /0/100/1f.2                storage        7 Series Chipset Family 6-port SATA Controller [AHCI mode]
    /0/2           scsi2       storage        
    /0/2/0.0.0     /dev/cdrom  disk           DVD-RAM UJ8E2

Smartcard reader name: `ACR1252U USB NFC Reader III`
Smartcard reader manufacturer: `ACS`
Smartcard: STARCOS version 3.5 from G&D (NFC smartcard)

Some logs when I try to perform binary update of a file: 
Logs: [log.txt](https://github.com/acshk/acsccid/files/2724472/log.txt)
GDB backtrace: [gdb_backtrace.txt](https://github.com/acshk/acsccid/files/2724495/gdb_backtrace.txt)

Additionally the exception that happens on code:

    javax.smartcardio.CardException: sun.security.smartcardio.PCSCException: SCARD_E_NOT_TRANSACTED
	at java.smartcardio/sun.security.smartcardio.ChannelImpl.doTransmit(ChannelImpl.java:219)
	at java.smartcardio/sun.security.smartcardio.ChannelImpl.transmit(ChannelImpl.java:89)
	at com.multicert.project.chipenc.apdubuilder.cardcomm.CardCommTest.testUpdateFileBinary(CardCommTest.java:196)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.InvokeMethod.evaluate(InvokeMethod.java:17)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:137)
	at com.intellij.junit4.JUnit4IdeaTestRunner.startRunnerWithArgs(JUnit4IdeaTestRunner.java:68)
	at com.intellij.rt.execution.junit.IdeaTestRunner$Repeater.startRunnerWithArgs(IdeaTestRunner.java:47)
	at com.intellij.rt.execution.junit.JUnitStarter.prepareStreamsAndStart(JUnitStarter.java:242)
	at com.intellij.rt.execution.junit.JUnitStarter.main(JUnitStarter.java:70)
    Caused by: sun.security.smartcardio.PCSCException: SCARD_E_NOT_TRANSACTED
	at java.smartcardio/sun.security.smartcardio.PCSC.SCardTransmit(Native Method)
	at java.smartcardio/sun.security.smartcardio.ChannelImpl.doTransmit(ChannelImpl.java:189)
	... 24 more

Also I should note, **if it's a normal-sized APDU (not extended), it works.**

Is this an issue?

Cheers.

godfreychung, 2019-01-03:
Yes, it seems to be an issue.
Can you provide the firmware version of the reader?

amportugal, 2019-01-03:
The current firmware version is:

    Firmware version: 41 43 52 31 32 35 32 55 5F 56 31 30 37

godfreychung, 2019-01-04:
According to my colleague, the reader returns 0xC3 if it receives invalid block from the card. It may be firmware or card issue. Would you mind sending a sample card to us for reproducing the problem?

closed 2020-03-26 by godfreychung