acshk/acsccid #16: Failed to transmit with protocol T0

roaldnefs, 2019-06-09:
I'm currently running into the following error when : `smartcard.Exceptions.CardConnectionException: Failed to transmit with protocol T0. Card protocol mismatch.` on Ubuntu 19.04 with the ACR122U-A9.

## Setup

```console
$ sudo apt-get install pcscd pcsc-tools libacsccid1
$ sudo modprobe -r pn533_usb pn533 nfc
$ sudo service pcscd restart
$ sudo pip install pyscard
```

Both `pcsc_scan` and `nfc-list` work fine, but whenever I try to use `pyscard` I run into the following error: `smartcard.Exceptions.CardConnectionException: Failed to transmit with protocol T0. Card protocol mismatch.`.

## Workaround

After uninstalling the official ACS drivers and restarting the `pcscd` service it works as expected, e.g.:
```console
# List the current installed drivers
$ dpkg --get-selections | grep libacsccid
libacsccid1

# Remove the driver
$ sudo dpkg -r libacsccid1

# Restart the pcscd service
$ sudo service pcscd restart
```

## More Information

More information about this issue is available at: https://github.com/LudovicRousseau/pyscard/issues/70.

godfreychung, 2019-06-10:
Can you use the following command to generate the log file?
`sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd | tee log.txt`

roaldnefs, 2019-06-10:
@godfreychung, here is the output of the command: `sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd`:

```
00000000 debuglog.c:299:DebugLogSetLevel() debug level=debug
00000241 configfile.l:284:DBGetReaderListDir() Parsing conf directory: /etc/reader.conf.d
00000039 configfile.l:321:DBGetReaderListDir() Skipping non regular file: .
00000006 configfile.l:360:DBGetReaderList() Parsing conf file: /etc/reader.conf.d/libccidtwin
00000079 configfile.l:321:DBGetReaderListDir() Skipping non regular file: ..
00000014 pcscdaemon.c:662:main() pcsc-lite 1.8.24 daemon ready.
00009631 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
00000146 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
00000147 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x0CF3, PID: 0xE300, path: /dev/bus/usb/001/002
00000142 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x0CF3, PID: 0xE300, path: /dev/bus/usb/001/002
00000131 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
00000140 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x04F3, PID: 0x20D0, path: /dev/bus/usb/001/003
00000130 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
00000180 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x0BDA, PID: 0x568B, path: /dev/bus/usb/001/004
00000145 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x0BDA, PID: 0x568B, path: /dev/bus/usb/001/004
00000222 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0003, path: /dev/bus/usb/002/001
00000376 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/003/001
00000158 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/003/001
00000170 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x05E3, PID: 0x0610, path: /dev/bus/usb/003/002
00000179 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x214B, PID: 0x7000, path: /dev/bus/usb/003/003
00000188 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x072F, PID: 0x2200, path: /dev/bus/usb/003/022
00000007 hotplug_libudev.c:436:HPAddDevice() Adding USB device: ACS ACR122U PICC Interface
00000046 readerfactory.c:1075:RFInitializeReader() Attempting startup of ACS ACR122U PICC Interface 00 00 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
00000236 readerfactory.c:950:RFBindFunctions() Loading IFD Handler 3.0
00000025 ifdhandler.c:1961:init_driver() Driver version: 1.4.30
00000681 ifdhandler.c:1978:init_driver() LogLevel: 0x0003
00000007 ifdhandler.c:1989:init_driver() DriverOptions: 0x0000
00000212 ifdhandler.c:2002:init_driver() LogLevel from LIBCCID_ifdLogLevel: 0x0007
00000005 ifdhandler.c:110:CreateChannelByNameOrChannel() Lun: 0, device: usb:072f/2200:libudev:0:/dev/bus/usb/003/022
00000004 ccid_usb.c:237:OpenUSBByName() Reader index: 0, Device: usb:072f/2200:libudev:0:/dev/bus/usb/003/022
00000007 ccid_usb.c:269:OpenUSBByName() interface_number: 0
00000002 ccid_usb.c:270:OpenUSBByName() usb bus/device: 3/22
00000002 ccid_usb.c:302:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Info.plist
00000656 ccid_usb.c:320:OpenUSBByName() ifdManufacturerString: Ludovic Rousseau (ludovic.rousseau@free.fr)
00000006 ccid_usb.c:321:OpenUSBByName() ifdProductString: Generic CCID driver
00000002 ccid_usb.c:322:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00007953 ccid_usb.c:406:OpenUSBByName() Try device: 3/22
00000013 ccid_usb.c:416:OpenUSBByName() vid/pid : 072F/2200
00000004 ccid_usb.c:483:OpenUSBByName() Checking device: 3/22
00000003 ccid_usb.c:554:OpenUSBByName() Trying to open USB bus/device: 3/22
00074012 ccid_usb.c:660:OpenUSBByName() Found Vendor/Product: 072F/2200 (ACS ACR122U PICC Interface)
00000042 ccid_usb.c:662:OpenUSBByName() Using USB bus/device: 3/22
00000017 ccid_usb.c:722:OpenUSBByName() bNumDataRatesSupported is 0
00023108 NotifySlotChange: 50 03 
00000036 -> 000000 65 00 00 00 00 00 00 00 00 00 
00000618 <- 000000 81 00 00 00 00 00 00 00 81 00 
00000026 -> 000000 65 00 00 00 00 00 01 00 00 00 
00000574 <- 000000 81 00 00 00 00 00 01 00 81 00 
00000030 ifdhandler.c:381:IFDHGetCapabilities() tag: 0xFB3, usb:072f/2200:libudev:0:/dev/bus/usb/003/022 (lun: 0)
00000008 readerfactory.c:396:RFAddReader() Using the reader polling thread
00000860 ifdhandler.c:381:IFDHGetCapabilities() tag: 0xFAE, usb:072f/2200:libudev:0:/dev/bus/usb/003/022 (lun: 0)
00000027 ifdhandler.c:476:IFDHGetCapabilities() Reader supports 1 slot(s)
00001003 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x214B, PID: 0x7000, path: /dev/bus/usb/003/003
00000705 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1038, PID: 0x1702, path: /dev/bus/usb/003/005
00000680 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1038, PID: 0x1702, path: /dev/bus/usb/003/005
00000654 hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x214B, PID: 0x7000, path: /dev/bus/usb/003/003
[... cut ...]
00006461 NotifySlotChange: 50 03 
00000645 ifdhandler.c:1154:IFDHPowerICC() action: PowerDown, usb:072f/2200:libudev:0:/dev/bus/usb/003/022 (lun: 0)
00000027 -> 000000 63 00 00 00 00 00 08 00 00 00 
00000514 <- 000000 81 00 00 00 00 00 08 00 81 00 
00000043 eventhandler.c:482:EHStatusHandlerThread() powerState: POWER_STATE_UNPOWERED
00030753 NotifySlotChange: 50 03
```

**Note:** this is without `libacsccid1` installed! Please let me know if you want the same output when I reinstall `libacsccid1`.

godfreychung, 2019-06-11:
Yes, please use `libacsccid1` to reproduce the problem. I want to know the log when your script prints out the error message `smartcard.Exceptions.CardConnectionException: Failed to transmit with protocol T0. Card protocol mismatch.`.

1. `sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd | tee log.txt`
2. Place a card on the reader.
3. Run your script which uses `pyscard`.

roaldnefs, 2019-06-11:
Here is the output with `libacsccid1` installed:

```bash
$ sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd
00000000 [140542259574720] debuglog.c:299:DebugLogSetLevel() debug level=debug
00000041 [140542259574720] utils.c:82:GetDaemonPid() Can't open /var/run/pcscd/pcscd.pid: No such file or directory
00000074 [140542259574720] configfile.l:284:DBGetReaderListDir() Parsing conf directory: /etc/reader.conf.d
00000013 [140542259574720] configfile.l:321:DBGetReaderListDir() Skipping non regular file: .
00000003 [140542259574720] configfile.l:360:DBGetReaderList() Parsing conf file: /etc/reader.conf.d/libccidtwin
00000035 [140542259574720] configfile.l:321:DBGetReaderListDir() Skipping non regular file: ..
00000006 [140542259574720] pcscdaemon.c:662:main() pcsc-lite 1.8.24 daemon ready.
00005336 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
00000171 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
00000154 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x0CF3, PID: 0xE300, path: /dev/bus/usb/001/002
00000126 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x0CF3, PID: 0xE300, path: /dev/bus/usb/001/002
00000124 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
00000135 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x04F3, PID: 0x20D0, path: /dev/bus/usb/001/003
00000104 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/001/001
00000123 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x0BDA, PID: 0x568B, path: /dev/bus/usb/001/004
00000118 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x0BDA, PID: 0x568B, path: /dev/bus/usb/001/004
00000179 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0003, path: /dev/bus/usb/002/001
00000396 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/003/001
00000163 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x1D6B, PID: 0x0002, path: /dev/bus/usb/003/001
00000148 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x05E3, PID: 0x0610, path: /dev/bus/usb/003/002
00000150 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x214B, PID: 0x7000, path: /dev/bus/usb/003/003
00000187 [140542259574720] hotplug_libudev.c:297:get_driver() Looking for a driver for VID: 0x072F, PID: 0x2200, path: /dev/bus/usb/003/005
00000007 [140542259574720] hotplug_libudev.c:436:HPAddDevice() Adding USB device: ACS ACR122U
00000051 [140542259574720] readerfactory.c:1075:RFInitializeReader() Attempting startup of ACS ACR122U 00 00 using /usr/lib/pcsc/drivers/ifd-acsccid.bundle/Contents/Linux/libacsccid.so
00000243 [140542259574720] readerfactory.c:950:RFBindFunctions() Loading IFD Handler 3.0
00000025 [140542259574720] ifdhandler.c:2930:init_driver() Driver version: 1.1.6
00000182 [140542259574720] ifdhandler.c:2947:init_driver() LogLevel: 0x0003
00000013 [140542259574720] ifdhandler.c:2958:init_driver() DriverOptions: 0x0000
00000006 [140542259574720] ifdhandler.c:2966:init_driver() ACSDriverOptions: 0x0003
00000005 [140542259574720] ifdhandler.c:2974:init_driver() ACR38CardVoltage: 0
00000011 [140542259574720] ifdhandler.c:2982:init_driver() ACR38CardType: 0
00000032 [140542259574720] ifdhandler.c:2995:init_driver() LogLevel from LIBCCID_ifdLogLevel: 0x0007
00000003 [140542259574720] ifdhandler.c:162:CreateChannelByNameOrChannel() Lun: 0, device: usb:072f/2200:libudev:0:/dev/bus/usb/003/005
00000005 [140542259574720] ccid_usb.c:263:OpenUSBByName() Reader index: 0, Device: usb:072f/2200:libudev:0:/dev/bus/usb/003/005
00000021 [140542259574720] ccid_usb.c:295:OpenUSBByName() interface_number: 0
00000004 [140542259574720] ccid_usb.c:296:OpenUSBByName() usb bus/device: 3/5
00000013 [140542259574720] ccid_usb.c:328:OpenUSBByName() Using: /usr/lib/pcsc/drivers/ifd-acsccid.bundle/Contents/Info.plist
00000173 [140542259574720] ccid_usb.c:346:OpenUSBByName() ifdManufacturerString: Advanced Card Systems Ltd.
00000005 [140542259574720] ccid_usb.c:347:OpenUSBByName() ifdProductString: ACS CCID driver
00000014 [140542259574720] ccid_usb.c:348:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00005333 [140542259574720] ccid_usb.c:432:OpenUSBByName() Try device: 3/5
00000008 [140542259574720] ccid_usb.c:442:OpenUSBByName() vid/pid : 072F/2200
00000002 [140542259574720] ccid_usb.c:526:OpenUSBByName() Checking device: 3/5
00000002 [140542259574720] ccid_usb.c:640:OpenUSBByName() Trying to open USB bus/device: 3/5
00066210 [140542259574720] ccid_usb.c:758:OpenUSBByName() Found Vendor/Product: 072F/2200 (ACS ACR122U)
00000019 [140542259574720] ccid_usb.c:760:OpenUSBByName() Using USB bus/device: 3/5
00000006 [140542259574720] ccid_usb.c:938:OpenUSBByName() bNumDataRatesSupported is 0
00000800 [140542259574720] -> 000000 63 00 00 00 00 00 00 00 00 00 
[... cut ...]
00000002 [140542215255808] winscard_svc.c:1063:MSGCleanupClient() Freeing SCONTEXT @0x563b27d8ad80
00003292 [140542223648512] winscard_svc.c:354:ContextThread() Client die: 14
00000028 [140542223648512] winscard.c:229:SCardReleaseContext() Releasing Context: 0x2EDC9306
00000011 [140542223648512] winscard_svc.c:1057:MSGCleanupClient() Thread is stopping: dwClientID=14, threadContext @0x563b27d8ac10
00000003 [140542223648512] winscard_svc.c:1063:MSGCleanupClient() Freeing SCONTEXT @0x563b27d8ac10
```

godfreychung, 2019-06-12:
```
00000004 [140542215255808] winscard.c:259:SCardConnect() Attempting Connect to ACS ACR122U 00 00 using protocol: 3
00000005 [140542215255808] readerfactory.c:821:RFReaderInfo() RefReader() count was: 1
00000003 [140542215255808] winscard.c:352:SCardConnect() powerState: POWER_STATE_IN_USE
00000002 [140542215255808] prothandler.c:108:PHSetProtocol() Attempting PTS to T=1
00000006 [140542215255808] ifdhandler.c:845:IFDHSetProtocolParameters() protocol T=1, usb:072f/2200:libudev:0:/dev/bus/usb/003/005 (lun: 0)
00000004 [140542215255808] ifdhandler.c:859:IFDHSetProtocolParameters() Timeout: 3000 ms
00000004 [140542215255808] winscard.c:431:SCardConnect() Active Protocol: T=1
00000004 [140542215255808] winscard.c:456:SCardConnect() hCard Identity: 74ea02aa
00000011 [140542215255808] winscard.c:518:SCardConnect() UnrefReader() count was: 2
00000005 [140542215255808] winscard_svc.c:511:ContextThread() CONNECT rv=0x0 for client 15
00000115 [140542215255808] winscard_svc.c:362:ContextThread() Received command: TRANSMIT from client 15
00000008 [140542215255808] readerfactory.c:848:RFReaderInfoById() RefReader() count was: 1
00000004 [140542215255808] winscard.c:1646:SCardTransmit() UnrefReader() count was: 2
00000004 [140542215255808] winscard_svc.c:683:ContextThread() TRANSMIT rv=0x8010000F for client 15
00002881 [140542215255808] winscard_svc.c:362:ContextThread() Received command: DISCONNECT from client 15
```

According to the log, your script uses `SCARD_PROTOCOL_T0 | SCARD_PROTOCOL_T1` (using protocol: 3) for card connection.

According to the ATR, It will choose T=1 for the highest protocol.
http://smartcard-atr.appspot.com/parse?ATR=3B8F8001804F0CA000000306030001000000006A

The active protocol is T=1. If your script passes T=0 to SCardTransmit(), it will return protocol mismatch error (0x8010000F).

closed 2020-03-26 by godfreychung