acshk/acsccid #17: ccid_usb.c:1316:ReadUSB() read failed (2/3): -7 LIBUSB_ERROR_TIME OUT

omaryesith, 2020-01-13:
Hi,

I'm having a ramdon trouble with my ACS ACR1281U smartcard reader, most of the time it works fine, but sometimes (after a reboot or a poweroff), the pcscd service does not see the smartcard reader.

root@vvsg-152-QA:~# systemctl status pcscd
● pcscd.service - PC/SC Smart Card Daemon
   Loaded: loaded (/lib/systemd/system/pcscd.service; indirect; vendor preset: enabled)
   Active: active (running) since Mon 2020-01-13 11:49:07 UTC; 16s ago
     Docs: man:pcscd(8)
 Main PID: 3430 (pcscd)
   CGroup: /system.slice/pcscd.service
           └─3430 /usr/sbin/pcscd --foreground --auto-exit
Jan 13 11:49:15 vvsg-152-QA pcscd[3430]: 01003612 ccid_usb.c:1316:ReadUSB() read failed (2/3): -7 LIBUSB_ERROR_TIME
OUT
Jan 13 11:49:16 vvsg-152-QA pcscd[3430]: 01003520 ccid_usb.c:1316:ReadUSB() read failed (2/3): -7 LIBUSB_ERROR_TIME
OUT
Jan 13 11:49:17 vvsg-152-QA pcscd[3430]: 01003479 ccid_usb.c:1316:ReadUSB() read failed (2/3): -7 LIBUSB_ERROR_TIME
OUT
Jan 13 11:49:18 vvsg-152-QA pcscd[3430]: 01003526 ccid_usb.c:1316:ReadUSB() read failed (2/3): -7 LIBUSB_ERROR_TIME
OUT
Jan 13 11:49:20 vvsg-152-QA pcscd[3430]: 01003479 ccid_usb.c:1316:ReadUSB() read failed (2/3): -7 LIBUSB_ERROR_TIME
OUT
Jan 13 11:49:21 vvsg-152-QA pcscd[3430]: 01003637 ccid_usb.c:1316:ReadUSB() read failed (2/3): -7 LIBUSB_ERROR_TIME
OUT
Jan 13 11:49:21 vvsg-152-QA pcscd[3430]: 00000016 ifdhandler.c:308:CreateChannelByNameOrChannel() failed
Jan 13 11:49:21 vvsg-152-QA pcscd[3430]: 00002734 readerfactory.c:1106:RFInitializeReader() Open Port 0x200000 Fail
ed (usb:072f/2224:libudev:0:/dev/bus/usb/002/003)
Jan 13 11:49:21 vvsg-152-QA pcscd[3430]: 00000014 readerfactory.c:376:RFAddReader() ACS ACR1281 1S Dual Reader init
 failed.
Jan 13 11:49:21 vvsg-152-QA pcscd[3430]: 00000072 hotplug_libudev.c:523:HPAddDevice() Failed adding USB device: ACS
 ACR1281 1S Dual Reader

godfreychung, 2020-01-14:
It seems that it is a firmware issue. Do you know the firmware version?

omaryesith, 2020-01-21:
No I don't know the firmware version and not sure about how to get it.

godfreychung, 2020-01-22:
Please refer to section 5.3.1 from API manual.
You need to send escape command "E0 00 00 18 00" to get firmware version.

omaryesith, 2020-01-22:
Hi!

This is the the firmware version: "ACR1281U_V520.00"

godfreychung, 2020-01-23:
Firmware V524 or later fixes the issue.
Please contact your retailer to obtain the firmware upgrade.

closed 2020-03-26 by godfreychung