acshk/acsccid #18: ACR39U-NF/UF issues with latest driver 1.1.8 (macOS)

golbiga, 2020-03-12:
I've started noticing the operating system (macOS 10.15.3) will stop recognizing my card when using the readers above and the latest drivers 1.1.8. Running pcsctest or reinserting the card will initialize the connection. At the time of the failures I see the following

```
info	2020-02-20 14:16:44.495419 -0500	com.apple.ifdbundle	06714564 ifdhandler.c:1640:IFDHTransmitToICC() ACS ACR39U ICC Reader (lun: 0)
default	2020-02-20 14:16:44.495486 -0500	com.apple.ifdbundle	00000077 openct/proto-t1.c:175:t1_transceive() T=1 state machine is DEAD. Reset the card first.
error	2020-02-20 14:16:44.495908 -0500	com.apple.ifdreader	ACS ACR39U ICC Reader: failed to transmit APDU
debug	2020-02-20 14:16:49.411067 -0500	pivtoken	ACS ACR39U ICC Reader: begin exclusive SmartCard session
info	2020-02-20 14:16:49.411994 -0500	com.apple.ifdbundle	04916270 ifdhandler.c:1640:IFDHTransmitToICC() ACS ACR39U ICC Reader (lun: 0)
default	2020-02-20 14:16:49.412066 -0500	com.apple.ifdbundle	00000081 openct/proto-t1.c:175:t1_transceive() T=1 state machine is DEAD. Reset the card first.
error	2020-02-20 14:16:49.412561 -0500	com.apple.ifdreader	ACS ACR39U ICC Reader: failed to transmit APDU
```

godfreychung, 2020-03-14:
The issue may be related to your card.
Do you know the ATR and APDUs to reproduce the issue?

golbiga, 2020-04-22:
Sorry it took so long, I believe this is what you need. 

ATR - 3B D6 97 00 81 B1 FE 45 1F 07 80 31 C1 52 11 18 F9
APDU - 00 20 00 80 08 32 36 38 37 31 33 35 31

godfreychung, 2020-04-23:
Thank you for your information.

To debug the issue, you need to modify `Info.plist` to enable the driver to output more debug messages.

Edit `/usr/local/libexec/SmartCardServices/drivers/ifd-acsccid.bundle/Contents/Info.plist` and change `ifdLogLevel` to `0x0007`.
```xml
<key>ifdLogLevel</key>
<string>0x0007</string>
```

Run **Console** application and enter `com.apple.ifdreader` to filter log messages. Please ensure that you have enabled **"Include Info Messages"** and **"Include Debug Messages"** from **Action** menu.

Restart `com.apple.ifdreader` using the following command:
```
killall -SIGKILL -m .*com.apple.ifdreader
```

You need to plug the reader again.

Run your PC/SC program to send the APDU.

Sorry, I don't have your card.
Can you provide the log for analysis?

golbiga, 2020-04-23:
Ok here is what I'm seeing in console after enabling logging as it just happened. 

I just ran a simple sudo command and it prompted for pin and then failed and prompted for password.

```
default	10:28:57.483557-0400	com.apple.ifdreader	APDU -> 00 20 00 80 08 32 36 38 37 31 33 35 31
debug	10:28:57.483805-0400	com.apple.ifdbundle	-> IFDHTransmitICC(00000000, proto=1, in=<private>)
info	10:28:57.483850-0400	com.apple.ifdbundle	94768086 ifdhandler.c:1640:IFDHTransmitToICC() ACS ACR39U ICC Reader (lun: 0)
debug	10:28:57.483910-0400	com.apple.ifdbundle	00000067 commands.c:2434:CmdXfrBlockTPDU_T1() T=1: 13 and 65535 bytes
default	10:28:57.483958-0400	com.apple.ifdbundle	00000049 openct/proto-t1.c:175:t1_transceive() T=1 state machine is DEAD. Reset the card first.
debug	10:28:57.483994-0400	com.apple.ifdbundle	<- IFDHTransmitICC(out=(null)) = 612
default	10:28:57.484445-0400	com.apple.ifdreader	APDU FAILED
error	10:28:57.484471-0400	com.apple.ifdreader	<private>: failed to transmit APDU
```

godfreychung, 2020-04-23:
Thank you for your log messages.

After tracing the source code (line number from the log), it seems that the previous APDUs (before APDU: 00 20 00 80 08 32 36 38 37 31 33 35 31) makes the card unresponsive.

Can you provide the log before the first line?

golbiga, 2020-04-27:
I'm starting to think this is a hardware issue with the reader. I used another reader all day on Friday without any APDU errors. I'll test again today and if nothing pops up then I know it's indeed a hardware issue and I will close the ticket.

godfreychung, 2020-04-27:
Ok. I'll wait for your result.

After reviewing T=1 protocol handling in the driver, I found that it does not handle ICC timeout properly. The driver sets the state to DEAD after receiving CCID error code 0xFE. If you see this error code in the log, it may be the reason why the operation cannot continue.

golbiga, 2020-04-29:
Ill take a look through the logs and see what I can dig up. Thanks!

closed 2023-04-03 by godfreychung