acshk/acsccid #21: ACR122U USB Timeout

vjardin, 2020-03-27:
Hi,

The ACR122U cannot be initialized by pcscd:
```
00000155 ccid_usb.c:347:OpenUSBByName() ifdManufacturerString: Advanced Card Systems Ltd.
00000010 ccid_usb.c:348:OpenUSBByName() ifdProductString: ACS CCID driver
00000002 ccid_usb.c:349:OpenUSBByName() Copyright: This driver is protected by terms of the GNU Lesser General Public License version 2.1, or (at your option) any later version.
00000787 ccid_usb.c:433:OpenUSBByName() Try device: 1/9
00000024 ccid_usb.c:443:OpenUSBByName() vid/pid : 072F/2200
00000002 ccid_usb.c:527:OpenUSBByName() Checking device: 1/9
00000001 ccid_usb.c:641:OpenUSBByName() Trying to open USB bus/device: 1/9
00068568 ccid_usb.c:754:OpenUSBByName() Found Vendor/Product: 072F/2200 (ACS ACR122U)
00000033 ccid_usb.c:756:OpenUSBByName() Using USB bus/device: 1/9
00000003 ccid_usb.c:934:OpenUSBByName() bNumDataRatesSupported is 0
00003938 -> 000000 63 00 00 00 00 00 00 00 00 00 
05001046 ccid_usb.c:1193:WriteUSB() write failed (1/9): -7 LIBUSB_ERROR_TIMEOUT
00000159 -> 000000 63 00 00 00 00 00 01 00 00 00 
05000634 ccid_usb.c:1193:WriteUSB() write failed (1/9): -7 LIBUSB_ERROR_TIMEOUT
00000036 -> 000000 63 00 00 00 00 00 02 00 00 00 
```

what kind of logs could be useful for such issue ?

godfreychung, 2020-03-28:
The log showed that it was a USB communication problem. The problem may be related to USB host controller or reader firmware.

Can you provide your platform information (e.g. hardware, OS, ...)?

LudovicRousseau, 2020-06-01:
@vjardin using HyperV virtualisation software again?

closed 2023-04-03 by godfreychung