acshk/acsccid #23: ACS ACR1251 1S CL Reader multiple errors and unstable

shaohme, 2020-08-25:
I'm experimenting with a ACR1251 reader and Yubikey 5C keys on my Debian bullseye/sid machine. I wanted to make the ACR1251 my primary NFC reader for GPG and PIV instead of using USB. Along with pcscd and opensc I have also installed libacsccid1.

Besides the usual exclusivity problems with GPG's scdaemon and pcscd, I seem to have stability problems. The ACR1251 beeps all of a sudden after having been plugged in for some time without having anything to read, and I don't refer to the initial beep the reader makes when just plugged in. 
Sometimes the reader just stops responding, and I have to re-plug it in again to make it responsive.
journalctl for pcscd unit prints alot of errors.

The lines below sometimes gets repeatedly printed until I unplug the device or cancel the PKCS#11 operating, like sudo,ssh,login,etc.
....
Aug 25 06:58:47 mkjws pcscd[8924]: 00000002 winscard.c:1620:SCardTransmit() Card not transacted: 0x80100016
Aug 25 06:58:47 mkjws pcscd[8924]: 00000009 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
Aug 25 06:58:47 mkjws pcscd[8924]: 00002301 commands.c:1608:CCID_Receive Activity aborted by Host
Aug 25 06:58:47 mkjws pcscd[8924]: 00000003 winscard.c:1620:SCardTransmit() Card not transacted: 0x80100016
Aug 25 06:58:47 mkjws pcscd[8924]: 00000008 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
Aug 25 06:58:47 mkjws pcscd[8924]: 00011052 commands.c:1608:CCID_Receive Activity aborted by Host
Aug 25 06:58:47 mkjws pcscd[8924]: 00000003 winscard.c:1620:SCardTransmit() Card not transacted: 0x80100016
Aug 25 06:58:47 mkjws pcscd[8924]: 00000007 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
Aug 25 06:58:47 mkjws pcscd[8924]: 00002293 commands.c:1608:CCID_Receive Activity aborted by Host
Aug 25 06:58:47 mkjws pcscd[8924]: 00000003 winscard.c:1620:SCardTransmit() Card not transacted: 0x80100016
Aug 25 06:58:47 mkjws pcscd[8924]: 00000008 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
Aug 25 06:58:47 mkjws pcscd[8924]: 00002320 commands.c:1608:CCID_Receive Activity aborted by Host
Aug 25 06:58:47 mkjws pcscd[8924]: 00000002 winscard.c:1620:SCardTransmit() Card not transacted: 0x80100016
Aug 25 06:58:47 mkjws pcscd[8924]: 00000009 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
Aug 25 06:58:47 mkjws pcscd[8924]: 00002408 commands.c:1608:CCID_Receive Activity aborted by Host
Aug 25 06:58:47 mkjws pcscd[8924]: 00000003 winscard.c:1620:SCardTransmit() Card not transacted: 0x80100016
Aug 25 06:58:47 mkjws pcscd[8924]: 00000009 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
Aug 25 06:58:47 mkjws pcscd[8924]: 00010979 commands.c:1608:CCID_Receive Activity aborted by Host
Aug 25 06:58:47 mkjws pcscd[8924]: 00000003 winscard.c:1620:SCardTransmit() Card not transacted: 0x80100016
Aug 25 06:58:47 mkjws pcscd[8924]: 00000021 ifdwrapper.c:543:IFDTransmit() Card not transacted: 612
Aug 25 06:58:47 mkjws pcscd[8924]: 51921737 commands.c:1608:CCID_Receive Card absent or mute

These are produced when the reader is plugged in:
Aug 25 06:57:24 mkjws pcscd[8924]: 00000004 eventhandler.c:305:EHStatusHandlerThread() Error powering up card: 2148532246 0x80100016
Aug 25 06:57:24 mkjws pcscd[8924]: 00000010 ifdhandler.c:1511:IFDHPowerICC() PowerUp failed
Aug 25 06:57:24 mkjws pcscd[8924]: 00298874 commands.c:254:CmdPowerOn Card absent or mute
Aug 25 06:57:23 mkjws pcscd[8924]: 10148529 commands.c:254:CmdPowerOn Card absent or mute

Do I have buggy hardware/firmware? If so, can the firmware perhaps easily be updated?

godfreychung, 2020-08-25:
Can you generate more detailed log using the following command?
```
sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd | tee log.txt
```
Please make sure that you have stopped `pcscd` before running the command.

Can you provide the firmware version?

shaohme, 2020-08-25:
> Can you generate more detailed log using the following command?
> 
> ```
> sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd | tee log.txt
> ```
> 
> Please make sure that you have stopped `pcscd` before running the command.
> 
> Can you provide the firmware version?

How do I get the firmware version?

shaohme, 2020-08-25:
> Can you generate more detailed log using the following command?
> 
> ```
> sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd | tee log.txt
> ```
> 
> Please make sure that you have stopped `pcscd` before running the command.
> 
> Can you provide the firmware version?

Here is a more verbose log. This time no sudden beeping by still same "Card not transacted: 612" etc. errors
[pcsc-error-log.txt](https://github.com/acshk/acsccid/files/5122915/pcsc-error-log.txt)

godfreychung, 2020-08-26:
> 
> 
> > Can you generate more detailed log using the following command?
> > ```
> > sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd | tee log.txt
> > ```
> > 
> > 
> > Please make sure that you have stopped `pcscd` before running the command.
> > Can you provide the firmware version?
> 
> How do I get the firmware version?

You need to send the escape command to get the firmware version.
Please refer to section 5.4.1 from [API manual](https://www.acs.com.hk/download-manual/4414/API-ACR1251U-1.13.pdf).

shaohme, 2020-08-28:
> Can you generate more detailed log using the following command?
> 
> ```
> sudo LIBCCID_ifdLogLevel=0x0007 pcscd -fd | tee log.txt
> ```
> 
> Please make sure that you have stopped `pcscd` before running the command.
> 
> Can you provide the firmware version?

Running this script ...
```python

#! /usr/bin/env python

from smartcard.scard import *
import smartcard.util

CMD = [ 0xE0, 0x00, 0x00, 0x18, 0x00 ]

try:
    hresult, hcontext = SCardEstablishContext(SCARD_SCOPE_USER)
    if hresult != SCARD_S_SUCCESS:
        raise Exception('Failed to establish context : ' +
            SCardGetErrorMessage(hresult))
    print('Context established!')

    try:
        hresult, readers = SCardListReaders(hcontext, [])
        if hresult != SCARD_S_SUCCESS:
            raise Exception('Failed to list readers: ' +
                SCardGetErrorMessage(hresult))
        print('PCSC Readers:', readers)

        if len(readers) < 1:
            raise Exception('No smart card readers')

        reader = readers[0]
        print("Using reader:", reader)
        try:
            hresult, hcard, dwActiveProtocol = SCardConnect(hcontext, reader,
                SCARD_SHARE_SHARED, SCARD_PROTOCOL_T0 | SCARD_PROTOCOL_T1)
            if hresult != SCARD_S_SUCCESS:
                raise Exception('Unable to connect: ' +
                    SCardGetErrorMessage(hresult))
            print('Connected with active protocol', dwActiveProtocol)

            try:
                hresult, response = SCardControl(hcard, SCARD_CTL_CODE(3500), CMD)
                print('RES: %s' % (response))
                if hresult != SCARD_S_SUCCESS:
                    raise Exception('Failed to control: ' + SCardGetErrorMessage(hresult))
            finally:
                hresult = SCardDisconnect(hcard, SCARD_UNPOWER_CARD)
                if hresult != SCARD_S_SUCCESS:
                    raise Exception('Failed to disconnect: ' +
                        SCardGetErrorMessage(hresult))
                print('Disconnected')

        except Exception as message:
            print("Exception:", message)

    finally:
        hresult = SCardReleaseContext(hcontext)
        if hresult != SCARD_S_SUCCESS:
            raise Exception('Failed to release context: ' +
                    SCardGetErrorMessage(hresult))
        print('Released context.')

except Exception as message:
    print("Exception:", message)
```
.. I get this output

```
Context established!
PCSC Readers: ['ACS ACR1251 Dual Reader [ACR1251 1S CL Reader PICC] 00 00', 'ACS ACR1251 Dual Reader [ACR1251 1S CL Reader SAM] 01 00']
Using reader: ACS ACR1251 Dual Reader [ACR1251 1S CL Reader PICC] 00 00
Connected with active protocol 2
RES: []
Disconnected
Exception: Failed to control: Feature not supported.
Released context.
```
Hope I did at least something right :|

godfreychung, 2020-08-28:
According to the output `ACS ACR1251 Dual Reader [ACR1251 1S CL Reader PICC] 00 00` of your script, you are using `ccid` driver instead of `acsccid` driver. Therefore, `SCARD_CTL_CODE(3500)` is not supported.

I think that you can remove `ccid` driver temporarily.

shaohme, 2020-08-28:
> According to the output `ACS ACR1251 Dual Reader [ACR1251 1S CL Reader PICC] 00 00` of your script, you are using `ccid` driver instead of `acsccid` driver. Therefore, `SCARD_CTL_CODE(3500)` is not supported.
> 
> I think that you can remove `ccid` driver temporarily.

After a few fixes I got version: ACR1251U_V215.0
Output from script:
```
Context established!
PCSC Readers: ['ACS ACR1251 1S CL Reader [ACR1251 1S CL Reader PICC] 00 00', 'ACS ACR1251 1S CL Reader [ACR1251 1S CL Reader SAM] 01 00']
Using reader: ACS ACR1251 1S CL Reader [ACR1251 1S CL Reader PICC] 00 00
Connected with active protocol 2
Version: ['á', '\x00', '\x00', '\x00', '\x0f', 'A', 'C', 'R', '1', '2', '5', '1', 'U', '_', 'V', '2', '1', '5', '.', '0']
Disconnected
Released context.
```

While doing this test, the reader beeped again after user after some time as described before. lsusb showed...
```
Bus 001 Device 031: ID 072f:221a Advanced Card Systems, Ltd ACR1251U-A1
```
... and dmesg showed this:
```
[38065.964216] usb 1-9.2: new full-speed USB device number 31 using xhci_hcd
[38066.275371] usb 1-9.2: New USB device found, idVendor=072f, idProduct=221a, bcdDevice= 2.15
[38066.275373] usb 1-9.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[38066.275374] usb 1-9.2: Product: ACR1251 PICC Reader
[38066.275375] usb 1-9.2: Manufacturer: ACS
[38097.765850] usb 1-9.1: USB disconnect, device number 30 <------------- here is when it beeped
[38192.240286] usb 1-9.2: USB disconnect, device number 31 <------------- unplugged by me an plugged back in again
[38194.228520] usb 1-9.2: new full-speed USB device number 32 using xhci_hcd
[38194.538538] usb 1-9.2: New USB device found, idVendor=072f, idProduct=221a, bcdDevice= 2.15
[38194.538540] usb 1-9.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[38194.538541] usb 1-9.2: Product: ACR1251 PICC Reader
[38194.538542] usb 1-9.2: Manufacturer: ACS
```
Just to let you know.

godfreychung, 2020-08-31:
According to the log and the reader behavior, it seems to be card compatibility issue. We have a beta firmware and it may fix the issue. In order to obtain the beta firmware for testing, you need to contact our technical support.

shaohme, 2020-08-31:
> According to the log and the reader behavior, it seems to be card compatibility issue. We have a beta firmware and it may fix the issue. In order to obtain the beta firmware for testing, you need to contact our technical support.

So it would seem Yubikey 5C NFC is incompatible. 

By technical support, do you mean ACS technical support?

Have you got a compatibility list of supported cards/keys to use with ACS readers?

godfreychung, 2020-08-31:
Yes, it's ACS technical support.

According to the specification, it supports ISO 14443 Type A and B cards, MIFARE, FeliCa, and all 4 types of NFC (ISO/IEC 18092) tags. I think that you may get more detailed information from ACS technical support.

shaohme, 2020-09-04:
> Yes, it's ACS technical support.
> 
> According to the specification, it supports ISO 14443 Type A and B cards, MIFARE, FeliCa, and all 4 types of NFC (ISO/IEC 18092) tags. I think that you may get more detailed information from ACS technical support.

I have written them through their official channels. We will see what happens ...

closed 2023-04-03 by godfreychung