acshk/acsccid #26: ccid vs acsccid - miss probing

vjardin, 2020-12-20:
Hi,

using the acsccid from:
```
  https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/
```
it can probe properly an ACR33U-A1:
```
# pwd
/usr/lib/pcsc/drivers
# ls -la
drwxr-xr-x    3 root     root             0 Dec 20 12:02 .
drwxr-xr-x    3 root     root             0 Dec 19 19:52 ..
drwxr-xr-x    3 root     root             0 Dec 20 11:46 ifd-acsccid.bundle
# pcscd -a -f -d
00000000 [2012581132] debuglog.c:299:DebugLogSetLevel() debug level=debug
00001344 [2012581132] configfile.l:293:DBGetReaderListDir() Parsing conf directory: /etc/reader.conf.d
00000929 [2012581132] configfile.l:330:DBGetReaderListDir() Skipping non regular file: .
00000660 [2012581132] configfile.l:330:DBGetReaderListDir() Skipping non regular file: ..
00000833 [2012581132] pcscdaemon.c:663:main() pcsc-lite 1.9.0 daemon ready.
00013324 [2011548992] hotplug_libusb.c:538:HPAddHotPluggable() Adding USB device: 2:2:0
00000832 [2011548992] readerfactory.c:1075:RFInitializeReader() Attempting startup of ACS ACR33U-A1 3SAM ICC Reader 00 00 using /usr/lib/pcsc/drivers/ifd-acsccid.bundle/Contents/Linux/libacsccid.so
```

but if there are both ccid and acsccid, then the wrong one is selected:
```
# pwd
/usr/lib/pcsc/drivers
# ls -la
drwxr-xr-x    4 root     root             0 Dec 20 12:25 .
drwxr-xr-x    3 root     root             0 Dec 19 19:52 ..
drwxr-xr-x    3 root     root             0 Dec 20 11:46 ifd-acsccid.bundle
drwxr-xr-x    3 root     root             0 Dec 19 23:29 ifd-ccid.bundle
# pcscd -a -f -d
00000000 [2012568844] debuglog.c:299:DebugLogSetLevel() debug level=debug
00001353 [2012568844] configfile.l:293:DBGetReaderListDir() Parsing conf directory: /etc/reader.conf.d
00000948 [2012568844] configfile.l:330:DBGetReaderListDir() Skipping non regular file: .
00000648 [2012568844] configfile.l:330:DBGetReaderListDir() Skipping non regular file: ..
00000849 [2012568844] pcscdaemon.c:663:main() pcsc-lite 1.9.0 daemon ready.
00049813 [2011536704] hotplug_libusb.c:538:HPAddHotPluggable() Adding USB device: 2:2:0
00000890 [2011536704] readerfactory.c:1075:RFInitializeReader() Attempting startup of ACS ACR33 ICC Reader 00 00 using /usr/lib/pcsc/drivers/ifd-ccid.bundle/Contents/Linux/libccid.so
```

The same, but of a Ubuntu 20.04 works fine. The proper acsccid is used.

I am not a deep expert in this framework: what could be wrong ? Which traces should I gather in order to isole the root cause ?

thank you,

godfreychung, 2020-12-21:
https://github.com/LudovicRousseau/PCSC/blob/4f7bf5c9a3acb80d6d8dea7256376a759e270b57/src/hotplug_libusb.c#L166
According to pcsc-lite, it uses `readdir()` to return the directory name. The `readdir()` does not guarantee any ordering.

vjardin, 2020-12-21:
I agree with your analysis, but then, how to make sure that the ccid driver skips the ACS devices so only the acsccid ones get used ?

For instance, from the code your are pointing,
   https://github.com/LudovicRousseau/PCSC/blob/4f7bf5c9a3acb80d6d8dea7256376a759e270b57/src/hotplug_libusb.c#L206
I understand that only the acsccid driver should be loaded for the ACS devices, doesn't it ?

godfreychung, 2020-12-22:
It can use `scandir()` instead of `readdir()` so that the directory name is returned in alphanumeric order.

vjardin, 2020-12-22:
I am not sure that it is the "spirit" of the design from @LudovicRousseau ; I can try to post a pull request in order to get some feedbacks about it.

LudovicRousseau, 2020-12-23:
pcscd has a priority check. If the driver is defined as a class driver `CCIDCLASSDRIVER` then another driver for the same device wll have a higher priority.
https://github.com/LudovicRousseau/PCSC/blob/master/src/hotplug_libudev.c#L313

It looks like the acsccid driver is configured as NOT a class driver
https://github.com/acshk/acsccid/blob/master/configure.ac#L220

So it should work fine, if pcscd uses `hotplug_libudev`.

From the traces above I see that pcscd has been configured to use `hotplug_libusb.c` instead of `hotplug_libudev.c` (default choice).
- Maybe openwrt can use libudev?
- Or can propose a patch for hotplug_libusb?

vjardin, 2020-12-23:
Thanks @LudovicRousseau for the followup. `hotplug_libusb.c` has to be used because of the constraints from openwrt environment (small footprint). Do you have any tip in order to have a proper patch to start with ? I do not have much experience with this environment of pcscd/libusb.

LudovicRousseau, 2020-12-24:
I may try to add support of multi driver and priority in `hotplug_libusb.c`. But not before a few days.
I had a look at the source code and this change will involve "complex" changes.

If you want to do it yourself I suggest to examine how it is done in `hotplug_libudev.c`.

vjardin, 2020-12-29:
Yes please, I would be interested in performing some tests of your patches on openwrt using hotplug_libusb.c. ping me whenever you want.

LudovicRousseau, 2021-01-02:
This should now be fixed in https://github.com/LudovicRousseau/PCSC/commit/113f74a1d7189ced85038b890253f2f3b477c3f0

Please test this code and report if your problem is fixed or not.

closed 2023-04-03 by godfreychung