acshk/acsccid #27: ACR123U not detected in Ubuntu Focal Fossa

takeruadelbert, 2021-01-26:
Hi ACS team, I would like to seek some advices and solutions regarding my problem.
I have device ACR123U and using Ubuntu 20.04 LTS OS.
However, the PCSC driver seems not working properly.
I did download the driver from here `https://www.acs.com.hk/en/products/343/acr123u-intelligent-contactless-reader/`
and I couldn't find a driver for Ubuntu Focal Fossa (I found only up to 19.10 (Eoan).
I tried to install Eoan driver in my Focal Fossa and when I check PCSC service status it shows like this :

[image]

However when I type `lsusb` the OS seems to show the device correctly though
[image]

Is there any solutions for this?

godfreychung, 2021-01-26:
Can you try to install libacsccid1 package from Ubuntu?
i.e. `sudo apt-get install libacsccid1`

takeruadelbert, 2021-01-26:
@godfreychung I have already installed that lib anyway
here's the output :
[image]

Is there any other libs that I should install?

godfreychung, 2021-01-26:
It seems to be a firmware issue.
Please kindly contact ACS for technical support.

closed 2021-01-26 by takeruadelbert