acshk/acsccid #28: ACR122U libusb error

audunmg, 2021-03-27:
Trying to use ACR122U with pcscd and it throws these errors 31905 times:

00000000 [139783334524480] ccid_usb.c:1191:WriteUSB() write failed (1/33): -1 LIBUSB_ERROR_IO
00000144 [139783334524480] ifdwrapper.c:364:IFDStatusICC() Card not transacted: 612
00000014 [139783334524480] eventhandler.c:336:EHStatusHandlerThread() Error communicating to: ACS ACR122U (A02014090501) 00 00
00000075 [139783326131776] ccid_usb.c:1191:WriteUSB() write failed (1/33): -1 LIBUSB_ERROR_IO
00000063 [139783326131776] ifdwrapper.c:364:IFDStatusICC() Card not transacted: 612
00000017 [139783326131776] eventhandler.c:336:EHStatusHandlerThread() Error communicating to: ACS ACR122U (A02014090501) 00 01
00000052 [139783326131776] ccid_usb.c:1191:WriteUSB() write failed (1/33): -1 LIBUSB_ERROR_IO
00000042 [139783326131776] ifdwrapper.c:364:IFDStatusICC() Card not transacted: 612
00000019 [139783326131776] eventhandler.c:336:EHStatusHandlerThread() Error communicating to: ACS ACR122U (A02014090501) 00 01

It repeats and fills the logs pretty fast.

Using acsccid compiled from latest master on arch linux.

godfreychung, 2021-03-28:
According to the log, it is very strange that two ACR122U readers have same USB serial number "A02014090501". I remembered that ACR122U reader does not store this property.

Do you know the firmware version?

audunmg, 2021-03-28:
Sorry, I don't know how do i find the firmware version.
Do you know if it's possible to update it?

I only have one reader, but it shows up as two readers when it shows up:
```
$ pcsc_scan 
Using reader plug'n play mechanism
Scanning present readers...
0: ACS ACR122U (A02014090501) 00 00
1: ACS ACR122U (A02014090501) 00 01
```

godfreychung, 2021-03-28:
The ACR122U reader has only 1 slot and it seems that your reader is not a genuine ACR122U reader. According to the log, the reader does not accept any command and I think that you cannot send the command to get the firmware version.

Can you paste the output of the command `sudo lsusb -v`?

audunmg, 2021-03-28:
Here's the output of lsusb -v:

```
$ lsusb|grep ACR
Bus 001 Device 116: ID 072f:2200 Advanced Card Systems, Ltd ACR122U
$ sudo lsusb -v -d 072f:2200
Bus 001 Device 116: ID 072f:2200 Advanced Card Systems, Ltd ACR122U
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        16
  idVendor           0x072f Advanced Card Systems, Ltd
  idProduct          0x2200 ACR122U
  bcdDevice            0.00
  iManufacturer           1 ACS
  iProduct                2 ACR122U PICC Interface
  iSerial                 3 A02014090501
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005d
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass        11 Chip/SmartCard
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      ChipCard Interface Descriptor:
        bLength                54
        bDescriptorType        33
        bcdCCID              1.00
        nMaxSlotIndex           1
        bVoltageSupport         7  5.0V 3.0V 1.8V 
        dwProtocols             3  T=0 T=1
        dwDefaultClock       4000
        dwMaxiumumClock      4000
        bNumClockSupported      0
        dwDataRate           9946 bps
        dwMaxDataRate      318280 bps
        bNumDataRatesSupp.      0
        dwMaxIFSD             254
        dwSyncProtocols  00000007  2-wire 3-wire I2C
        dwMechanical     00000000 
        dwFeatures       000204BA
          Auto configuration based on ATR
          Auto voltage selection
          Auto clock change
          Auto baud rate change
          Auto PPS made by CCID
          Auto IFSD exchange
          Short APDU level exchange
        dwMaxCCIDMsgLen       271
        bClassGetResponse    echo
        bClassEnvelope       echo
        wlcdLayout           none
        bPINSupport             0 
        bMaxCCIDBusySlots       1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
```

godfreychung, 2021-03-29:
Comparing the value with the original ACR122U reader, I believe that your reader is fake. Please contact your retailer for refund.

audunmg, 2021-04-26:
Thank you for your time, it turned out to be fake.

closed 2021-04-26 by audunmg