acshk/acsccid #29: Slow card removal detection after loading authentication keys

hinxx, 2021-04-04:
I'm working with ACR122U on Linux. I'm using a separate thread to detect reader and card arrival / removal. The timeout argument to the `SCardGetStatusChange()`  is set to `INFINITE`.

When not performing any card access, I can see removal of the card being reported in less than 300 ms. As soon as I issue the command to load the keys the time increases to ~ 1.6 s. I'm looking at the timestamps in pcscd output.

Is this increase in card removal detection expected?

godfreychung, 2021-04-07:
Are you using the command "Load Authentication Keys" with MIFARE Classic card?

hinxx, 2021-04-07:
I think so. This is one of the cards I got with the reader, supposedly 1k MIFARE:
```
Card ATR: 3B 8F 80 01 80 4F 0C A0 00 00 03 06 03 00 01 00 00 00 00 6A
```

godfreychung, 2021-04-08:
When the application sends APDU to the card, the reader uses communication timeout to detect card removal. Actually, the communication timeout is longer than the polling timeout.

hinxx, 2021-04-14:
I managed to get an earlier card removal detection by issuing card ID read requests at a moderate rate (~200 ms) once the card is authenticated. This seems to tell the pcscd daemon (reader?) to issue a state change to the thread waiting such change. If there's a cleaner way, please, let me know.

closed 2023-04-03 by godfreychung