acshk/acsccid #31: Unresolved soname dependencies

linuxunderground, 2022-05-27:
Quality Assurance of Gentoo package manager says about acsccid-1.1.8 :
```
QA Notice: Unresolved soname dependencies:
  /usr/lib/readers/usb/ifd-acsccid.bundle/Contents/Linux/libacsccid.so: libusb-1.0.so.0
```
Gentoo package manager says the same for ccid-1.4.35 but not for ccid-1.5.0.
Therefore, could you consider making a new version of acsccid with files imported from ccid-1.5.0 instead of ccid-1.4.31.

godfreychung, 2022-05-30:
I compared `configure.ac` and `Makefile.am` from ccid-1.5.0 and I did not find any update.

Sorry, I'm not familiar with Gentoo packaging. Is it related to specify dependencies in the package file?

linuxunderground, 2022-05-30:
After installing ccid-1.5.0 (and reinstalling libusb) I can no longer reproduce this issue.
Sorry  for the noise.

closed 2022-05-30 by linuxunderground