acshk/acsccid #32: ACR 1281-u : Unable to load mifare key

Sukhoimaster, 2022-10-16:
Failed while using APDU on *ISO 14443-4 A* Card  ; Mifare load key `[0xFF, 0x82, 0x00, 0x20, 0x06] + hex(key)`

### Platform
- OS : Linux beaglebone 4.19.94-ti-r42 #1buster SMP PREEMPT Tue Mar 31 19:38:29 UTC 2020 armv7l GNU/

### Version 
- Reader driver ; `libacsccid1 v1.1.6-1` _installed from linux apt package manager_ 
- Middleware (PC/SC) ; `libpcsclite1:armhf  v1.8.24-1` _installed from linux apt package manager_
- USB Driver ; `libusb-1.0-0:armhf   v2:1.0.22-2` _installed from linux apt package manager_
- python3-pyscard ; `1.9.9-2build1-armhf` _installed from linux apt package manager_

> With error ; `Failed to transmit with protocol T1. Transaction failed.` using the [pyscard API](https://pyscard.sourceforge.io)
Force using T0_protocol were also _FAILED_

> The failed card ATR ;*(3B 89 80 01 D0 05 00 03 00 5E BB 00 00 3B)* Fetch via [ACS Quickview Tools](https://www.acs.com.hk/download-utility-tools/2097/ACS%20QuickView%20Tool%20v2.02.zip)
[image]

Failed process at `line 1146` [log_acr1281-u_failed.txt](https://github.com/acshk/acsccid/files/9795798/log_acr1281-u_failed.txt)

Below is the 
### Success sample

Using the same environment setup but with different card _(Mifare+ SL1_4k)_
> The success card ATR ; *(3B 8F 80 01 80 4F 0C A0 00 00 03 06 03 00 37 00 00 00 00 5C)*

[image]
Here's the succeed process log ; [log_acr1281-u.txt](https://github.com/acshk/acsccid/files/9795864/log_acr1281-u.txt)

### Another approach 
Using another environment setup (latest/stable release) as below  ;
-`acsccid ; 1.1.8`
-`pcscd ;  1.9.9`
-`libusb ; 1.0.26`
-`pyscard ; 2.0.5`

> Changes of environment setup has done with proper install & removal/uninstall step.
e.g: apt autoremove <package> , make uninstall , make distclean , configure.

> Also getting the same problem with the *ISO 14443-4 A* Card but *SUCCESS* with  _(Mifare+ SL1_4k)_ 

> I did also try replace acsccid with [ccid](https://ccid.apdu.fr/files/ccid-1.5.0.tar.bz2) `1.5.0' release from @LudovicRousseau 
- since the latest release is now supporting the ACR1281-U
- Resulting the same as above

godfreychung, 2022-10-17:
According to your card ATR (3B 89 80 01 D0 05 00 03 00 5E BB 00 00 3B), it is not a Mifare card. Therefore, the load key command cannot be used.

Sukhoimaster, 2022-10-17:
I was told the card is DESFIRE EV2 loaded with Mifare+ data and configuration, 
Any source of getting the supported APDU command. 
Since in the ACR1281-u API DOCUMENTATION only explaining very short about ISO 14443-4 A

godfreychung, 2022-10-17:
You need to contact NXP for the reference manual.

Sukhoimaster, 2022-10-17:
Okay , Thanks for your time

closed 2022-10-17 by Sukhoimaster