acshk/acsccid #35: Configuration errors with config/config.path files and config.h.in

temaps, 2023-10-23:
```
(%build): /bin/sh -e /var/tmp/rpm-tmp.knN8h6
+ umask 022
+ cd /tmp/abf/rpmbuild/BUILD
+ cd acsccid-1.1.10
+ '[' 1 -eq 1 ']'
+ '[' 1 -eq 1 ']'
+ find . -name config.guess -o -name config.sub
+ read i
+ '[' -e configure.ac -a -e Makefile.am ']'
+ find . -name configure.ac
+ xargs dirname
+ read D
+ pushd .
+ grep -qE '(LT_INIT|LIBTOOL)' configure.ac
/tmp/abf/rpmbuild/BUILD/acsccid-1.1.10 /tmp/abf/rpmbuild/BUILD/acsccid-1.1.10
+ libtoolize --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'config'.
libtoolize: linking file 'config/ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
++ grep -v '^.$'
++ find . -name '*.m4'
++ sort
++ uniq
++ grep -vE 'ac(local|include).m4'
++ xargs dirname
++ cut -d/ -f2-
++ read R
++ '[' -e m4/configure.ac ']'
++ echo -n '-Im4 '
++ read R
+ aclocal -Im4
+ automake -a --foreign
configure.ac:36: installing 'config/ar-lib'
configure.ac:29: installing 'config/compile'
configure.ac:20: installing 'config/config.guess'
configure.ac:69: error: required file 'config/config.rpath' not found
configure.ac:20: installing 'config/config.sub'
configure.ac:10: installing 'config/install-sh'
configure.ac:10: installing 'config/missing'
configure.ac:23: error: required file 'config.h.in' not found
src/Makefile.am: installing 'config/depcomp'
configure.ac: installing 'config/ylwrap'
```

temaps, 2023-10-23:
At what point should the bootstrap script work?
```
#
# Copy config.rpath to the directory "config".
# This file is needed by iconv.m4.
#
if [ -f /usr/share/gettext/config.rpath ]; then
	if [ ! -d config ]; then
		mkdir config
	fi
	cp /usr/share/gettext/config.rpath config
else
	echo "Please install gettext before running this script."
	exit 1
fi
```

temaps, 2023-10-27:
> It seems that you are building RPM package. Please refer to https://src.fedoraproject.org/rpms/pcsc-lite-acsccid/blob/rawhide/f/pcsc-lite-acsccid.spec It calls ./bootstrap from %build.

Yes, thank you very much. I eventually did so. I only have a question about the rules file. It is not in spec fedora:
https://abf.io/import/acsccid/blob/rosa2023.1/acsccid.spec#lc-43

godfreychung, 2023-10-27:
It seems that it is specific to Fedora.
See https://src.fedoraproject.org/rpms/pcsc-lite-ccid/c/e889690138e07a5fc81a6aa57911e960b47376ac?branch=d97f8f84ea8de1db841e87f38e135f821ff2499b

closed 2024-02-15 by godfreychung